GrumpyOldPizza/ArduinoCore-stm32wb #1: WB55 loss of Advertise

Defragster, 2021-03-22:
Early observations with Firefly and Katydid are seeing occasional failure to Advertise/present their MAC to pick up with "STM BLE Sensor" (SBS app) on Android Phone.

Starting with sketch : BlueST_Console.ino where loop() modified to:
`uint32_t lCnt = 0;
uint32_t msSec = millis();
void loop()
{
  int c;
  lCnt++;
  if ( millis() - msSec > 2000 ) {
    msSec = millis();
    Serial.print( "Cnt of ");
    lCnt /= 2;
    Serial.println( lCnt);
    SerialBLE.print("\nLoopCount = ");
    SerialBLE.println(lCnt);
    lCnt = 0;

    if (!BLE.advertising() && !BLE.connected()) {
      BLE.advertise();
    }

    SerialBLE.print("Temperature = ");
    SerialBLE.print(STM32WB.readTemperature());
    SerialBLE.println(" *C");

    SerialBLE.print("Battery = ");
    SerialBLE.print(STM32WB.readBattery());
    SerialBLE.println(" V");

    SerialBLE.println();

    while ((c = SerialBLE.read()) >= 0) {
      if ( c == '1' )
        digitalWrite(LED_BUILTIN, HIGH);
      if ( c == '0' )
        digitalWrite(LED_BUILTIN, LOW);

      if (c == '\n') {
        Serial.write('\r');
      } Serial.write(c);
    }
  }
}
`

Often it works after upload and the SBS app will see devices to connect. On some occasions it does not, and a board 'Reset' does not seem to be enough to restore connectivity?  That is on Katydid and power switch cycle will then show the device.

When the SBS app works the MAC is shown and the device can be selected and opening the " ... / Open Debug Console " with display the recurring 2 seconds messages sent with SerialBLE.print()

On occasion that Console will stop showing the incoming .print() items. Once connected the connect seems 'sticky' and assuming this behavior is similar to the failure to present in the first place?  It may be a second instance.

It seems that in these cases going to the 'Bottom Red Line' that is Debug OUTPUT and sending '1' or '0' the above code still responds and will as coded turn the LED On or Off. And it seems at that point the .print() data lines can return to the Console.

GrumpyOldPizza, 2021-03-24:
What device is used on the other side ? Android, Linux or iOS ?

There are some issues with some peers reconnecting automatically on Firefly reset. One way to work around that is to use a RANDOM STATIC address:

BLE.begin(23, BLE_OPTION_RANDOM_STATIC_ADDRESS);

Can you recheck on the "does not show up on ST BLE Sensor" with "nRF Connect" ?

Defragster, 2021-03-25:
Device :: Note 20 Android Phone running the "ST BLE Sensor app".
I made an issue because when I noted it to Kris he said I should as it was like something seen.  It was intermittent  at best - and just starting so thought it might be me messing with the code.

I've edited the .begin(...RANDOM...) into that sketch for when I can get back to it. That sketch is using "BLE_UART_PROTOCOL_BLUEST" that works with above ST BLE app.

I've changed sketches to a BMA400 one just now - using NORDIC and AdaF 'BlueFruit Connect' for that sketch and it seems more reliable - though I'm still working out the BMA400 interrupt and having the sketch not sleep but trying to get 400 Hz continuous samples - so mostly watching the SerMon ... and edit/rebuild/...