GrumpyOldPizza/ArduinoCore-stm32wb #2: BLE Central Implementation

BigCorvus, 2022-09-02:
Hi,
is there a way of implementing BLE central functionality ideally concurrently with peripheral role  on your core? I have a use case that requres a connection to multiple BLE peripherals while acting as a peripheral at the same time. I'd love to see how the WB55 performs on this task.
Keep up the great work!
Best,
Arthur

nguyenmanhthao996tn, 2023-01-27:
I have the same question! I just need to scan the available devices and get their ID. 
I tried to use examples from BLE folder with following steps:

- Set the role to GAP_PERIPHERAL_ROLE | GAP_CENTRAL_ROLE
- Enable these event masks: **ACI_GAP_PROC_COMPLETE_EVENT**, **HCI_LE_ADVERTISING_REPORT_EVENT**, **HCI_LE_EXTENDED_ADVERTISING_REPORT_EVENT**
- Call the function named _aci_gap_start_general_discovery_proc_ to trigger the scan

The _aci_gap_start_general_discovery_proc_ function seems to work. Because it's always return **BLE_STATUS_NOT_ALLOWED** before the scan timeout (10.24 seconds) and **BLE_STATUS_SUCCESS** after the scan!

However, the events that used to retrieve data are never called! Don't know if I miss something. Tried to check with the stack from STM32 but they always use polling + switch/case.

GrumpyOldPizza, 2023-02-04:
I need to add that hopefully soon than later. There are a few issues though. Main one is the API ... ArduinoBLE uses (I think) ::available() ... That just feels wrong. Scanned devices get discovered async, so how to allocate the backing object ...

The BLE class itself uses a switch statement to process it's events (BLELocalDevice::process()). So hooking in the ST style callback functions is not supported. I'd think it should be possible to write some small piece of code to bypass the whole BLE class and use ST's stack directly (raw event-callback and raw request functions).

BigCorvus, 2023-02-24:
Cool! 
Meh, BLE code without callbacks is like soldering without heat. Who needs the ArduinoBLE library anyway? ;)