GrumpyOldPizza/ArduinoCore-stm32wb #5: nothing does compile with arduino ide and arduino ide 2

hannescam, 2022-12-25:
[image]
these are the errors

GrumpyOldPizza, 2022-12-25:
Got a fix coming. For some reason some sequence of include files misses a key include file ...

hannescam, 2022-12-26:
when will the fix roll out? i dont want to presure you but i want to know when it will roll out because it will affect how i taccle my projects

maxgerhardt, 2022-12-27:
This core is uncompilable even in the latest stable 0.1.4 version beyond that error -- the simplest possible "Empty" example gives an error on `CDC_TX_PACKET_SIZE` not being found? Board = NUCLEO-WB55RG, External Storage "None", CPU Speed: 64MHz, code

```cpp
void setup() {}

void loop() {}
```

```
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp: In member function 'virtual size_t Uart::write(const uint8_t*, size_t)':
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp:209:36: error: 'CDC_TX_PACKET_SIZE' was not declared in this scope
                     if (tx_count > CDC_TX_PACKET_SIZE) {
                                    ^~~~~~~~~~~~~~~~~~
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp:209:36: note: suggested alternative: 'UART_TX_PACKET_SIZE'
                     if (tx_count > CDC_TX_PACKET_SIZE) {
                                    ^~~~~~~~~~~~~~~~~~
                                    UART_TX_PACKET_SIZE
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp:273:28: error: 'CDC_TX_PACKET_SIZE' was not declared in this scope
             if (tx_count > CDC_TX_PACKET_SIZE) {
                            ^~~~~~~~~~~~~~~~~~
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp:273:28: note: suggested alternative: 'UART_TX_PACKET_SIZE'
             if (tx_count > CDC_TX_PACKET_SIZE) {
                            ^~~~~~~~~~~~~~~~~~
                            UART_TX_PACKET_SIZE
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp: In static member function 'static void Uart::transmitCallback(Uart*)':
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp:384:28: error: 'CDC_TX_PACKET_SIZE' was not declared in this scope
             if (tx_count > CDC_TX_PACKET_SIZE) {
                            ^~~~~~~~~~~~~~~~~~
"C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -march=armv7e-m -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -mfloat-abi=hard -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions -MMD -DARDUINO=10819 -DARDUINO_STM32WB_NUCLEO -DARDUINO_ARCH_STM32WB -DSTM32WB55xx -fsingle-precision-constant -D__SYSTEM_CORE_CLOCK__=64000000L -DSTORAGE_TYPE=0 "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/CMSIS/Core/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/CMSIS/DSP/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/CMSIS/Device/ST/STM32WBxx/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/STM32WBxx/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4\\cores\\arduino" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4\\variants\\NUCLEO-WB55RG" "C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4\\cores\\arduino\\main.cpp" -o "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_244134\\core\\main.cpp.o"
"C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -march=armv7e-m -mfpu=fpv4-sp-d16 -mthumb -mabi=aapcs -mfloat-abi=hard -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib -fno-rtti -fno-exceptions -MMD -DARDUINO=10819 -DARDUINO_STM32WB_NUCLEO -DARDUINO_ARCH_STM32WB -DSTM32WB55xx -fsingle-precision-constant -D__SYSTEM_CORE_CLOCK__=64000000L -DSTORAGE_TYPE=0 "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/CMSIS/Core/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/CMSIS/DSP/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/CMSIS/Device/ST/STM32WBxx/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4/system/STM32WBxx/Include" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4\\cores\\arduino" "-IC:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4\\variants\\NUCLEO-WB55RG" "C:\\Users\\Max\\AppData\\Local\\Arduino15\\packages\\TleraCorp\\hardware\\stm32wb\\0.1.4\\cores\\arduino\\new.cpp" -o "C:\\Users\\Max\\AppData\\Local\\Temp\\arduino_build_244134\\core\\new.cpp.o"
C:\Users\Max\AppData\Local\Arduino15\packages\TleraCorp\hardware\stm32wb\0.1.4\cores\arduino\Uart.cpp:384:28: note: suggested alternative: 'UART_TX_PACKET_SIZE'
             if (tx_count > CDC_TX_PACKET_SIZE) {
                            ^~~~~~~~~~~~~~~~~~
                            UART_TX_PACKET_SIZE
exit status 1
Fehler beim Kompilieren für das Board NUCLEO-WB55RG.
```

GrumpyOldPizza, 2022-12-27:
Yup. Problem is that I never cross checked with the Nucleo board, which is the only one not having USB.

hannescam, 2022-12-27:
@GrumpyOldPizza it has usb one is stlink and the other is the stm32

GrumpyOldPizza, 2022-12-27:
The USB on ST-Link is a UART from STM32WB's point of view.

hannescam, 2022-12-27:
yes thats right but there is a second usb thats directly connected to the stm32wb55rg

GrumpyOldPizza, 2022-12-27:
Correct. But that USB connector is not used. So I missed all the typos and cooy%paste errors that assumed a CDC class is used.

hannescam, 2022-12-27:
and there is also a typo in the hardwareserial.cpp file and i opend allready a pull request

hannescam, 2022-12-27:
i also made a quick and dirty fix and that was with just replacing all the cdc things with uart

GrumpyOldPizza, 2022-12-27:
Obviously. The other one is to add in include for "Rtos_api.h" into armv7m.h.

GrumpyOldPizza, 2023-02-04:
Are all of those issues fixed now ?