GrumpyOldPizza/ArduinoCore-stm32wb #8: reading any other analog pins than a0-a5 dont work

hannescam, 2023-01-05:
reading any other analog pins than a0-a5 return just 0 but with this code it works on the same pins: 
```
stm32wb_gpio_pin_configure(STM32WB_GPIO_PIN_PA4, (STM32WB_GPIO_PUPD_NONE | STM32WB_GPIO_MODE_ANALOG));
input = __analogReadChannel((STM32WB_ADC_CHANNEL_1 + ADC_CHANNEL_9), 2 /* __analog_readPeriod */);
```

maxgerhardt, 2023-01-05:
To be more clear: The Nucleo has more ADC capabable pins which are not defined / marked correctly in the `variant.cpp`, e.g PA4 ("D10") being ADC1_Channel9, so the user can't do an `analogRead()` on them although it could be made possible.

hannescam, 2023-01-05:
i know but could you make it possible through analogread()

GrumpyOldPizza, 2023-01-06:
The NUCLEO variant is setup to handle excatly the official documented pin functions, which is D0-D13, SDA/SCL on D14/D15 and A0-A5, along with the PWMs that ST documents and that X-Nucleo boards are designed to make use of.

There are some more complications coming from the API in which D0-D5 cannot be analog inputs and such ...

So if there is a hard requirement to have more analogic inputs (other than just for cosmetic reasons), you can always add them in your forked tree.

closed 2023-01-06 by GrumpyOldPizza

maxgerhardt, 2023-01-06:
>D15 and A0-A5, along with the PWMs that ST documents and that X-Nucleo boards are designed to make use of.

STMicroelectronic's official Arduino core disagrees with this design decision to make the ADC pins which aren't placed on the Arduino-compatible header unavailable, the PA4 pin is available there as `A10`.

https://github.com/stm32duino/Arduino_Core_STM32/blob/294ace532b0638ab39ffcb929b0c5e9ba9c20c6c/variants/STM32WBxx/WB55R(C-E-G)V/variant_P_NUCLEO_WB55RG.cpp#L75-L93