GrumpyOldPizza/ArduinoCore-stm32wb #14: STM32WB55 Nucleo, PDM library error 'SERIAL_STATUS_BUSY' was not declared in this scope

federicaventriglia, 2025-01-23:
Hi everyone, maybe it's just me but I'm trying to use this library to run the PDM example on a P-NUCLEO WB55 connected to to 2 PDM microphones but when I try to build the normal example I get these compilation errors:

```
_SerialUSB.ino:32:55: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'void (*)()' [-fpermissive] 
        Serial.write(&data[0], PDM_BUFFER_SIZE, status);
```

and also

```
 error: 'SERIAL_STATUS_BUSY' was not declared in this scope
         while (status == SERIAL_STATUS_BUSY)
```

am I doing something wrong?

Also, I can't seem to get the USB Serial device to show up, even for just a simple Serial.prinln(), I've worked with the official core from ST up until this moment so I might be a little new to this one, apologies for the questions.

kriswiner, 2025-01-23:
I think you will need to use the Arduino core in this repo in order to get
any examples to work with it...Or comment out the offending lines, but then
you won't have USB Serial...or use the official core and their PDM
example...

In other words, these examples are not necessarily compatible with the
official ST core. That is because the pin out is different, and we have
written our own drivers for most of the serial peripherals including USB
Serial, etc.

federicaventriglia, 2025-01-23:
Ah thank you for clarifying that. It makes sense! Although the official core doesn’t have a PDM library I believe, that’s why I was interested about this one. Do you think trying to implement your library for the official core would be difficult?

kriswiner, 2025-01-23:
"Do you think trying to implement your library for the official core would
be difficult?"

Probably...why not just use the Arduino core and the Arduino IDE? I know
this works...

federicaventriglia, 2025-01-24:
I have a STM32 core unfortunately

kriswiner, 2025-01-24:
??

This <https://github.com/GrumpyOldPizza/ArduinoCore-stm32wb> is an
excellent STM32WB5X Arduino core.

There is also an STM32WB55 dev board
<https://www.tindie.com/products/tleracorp/firefly-ble-stm32wb55-development-board/>
(Firefly) and an STM32WB5MMG dev board
<https://www.tindie.com/products/tleracorp/sasquatch-stm32wb5mmg-development-board/>
(Sasquatch)
available. Not sure if we have put the core for the latter on github yet
though...

Either of these would allow use of a PDM mic.

GrumpyOldPizza, 2025-02-06:
There is more to PDM internally than meets the eye. STM32Duino just does not have the infrastructure to make that remotely feasible. But feel free to give it a go. After all the source code is available.

Yes, I need to update the GitHub for the newer boards, and yes I should add USB Audio ...